q-pup/q-pup-mechanical #28: Consider set screw for shaft into rotor adaptor plate

matthew-reynolds, 2022-01-09:
> mild concern about the shaft deforming over time and introducing extra slop over time

Agreed, we could maybe put a set-screw into the rotor adaptor plate and an allen-key-sized hole in the motor casing? Or perhaps we should just see how it is for now.

_Originally posted by @matthew-reynolds in https://github.com/q-pup/q-pup-mechanical/issues/24#issuecomment-1008352812_

wmmc88, 2022-01-10:
I think for now just keep it as is, but keep it on our watchlist of stuff that may need iteration

matthew-reynolds, 2022-01-26:
Based on how the hex shaft was fitting in the torque test rotor adaptor, I'm going to close this for now. It can be re-opened if needed later.

closed 2022-01-26 by matthew-reynolds