q-pup/q-pup-mechanical #33: Retolerance Parts to new spec

wmmc88, 2022-01-19:
Set tolerances to some number between 0.075 and 0.1mm

Using the [v2 makers muse tolerance guage](https://www.makersmuse.com/clearance-and-tolerance-3d-printer-gauge)

[image]

wraftus, 2022-01-20:
## Features with tolerance
- [ ] hex bore in [`rotor_adapter`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/rotor_adaptor.SLDPRT)
- [ ] hex bore in [`cam_a`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/cam_a.SLDPRT)
- [ ] hex bore in [`cam_b`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/cam_b.SLDPRT)
- [ ] bearing bore and pitch radius in [`cycloidal_disk`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/cycloidal_disk.SLDPRT)
- [ ]  bearing bore in [`motor_casing`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/motor_casing.SLDPRT)
- [ ] pin bore and in [`output_shaft`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/output_shaft.SLDPRT)
- [ ] ??? in [`rotor_adapter_ring`](https://github.com/q-pup/q-pup-mechanical/blob/main/q-pup/motor_module/rotor_adaptor_ring.SLDPRT)

matthew-reynolds, 2022-01-20:
We shouldn't need a list of things, if you just update the tolerances file everything will update

wmmc88, 2022-01-20:
I think @wraftus mightve wanted to do a sanity check that those things are all toleranced

PR #74 (2022-02-06): Update tols
Closes #33

Based on #70, we found:
 - 7.95mm dia shaft slip fits in 8.01mm hole
 - 6.8532mm flat-to-flat hex slip fits in 6.8732mm flat-to-flat hole

Tols are therefore set accordingly:
 - `tol_dia_id_slip=tol_dia_od_slip=(8.01-7.95)/2=0.03mm`
 - `tol_dia_general_slip=(6.8732-6.8532)/4=0.05mm`
 
[image]
[image]

See tolerances.txt for the full set of updated tolerances. Clearance and intfc fits were sorta just yoloed based off slip. Layer tols just set to match general X-Y tols.

Additionally, the pin dia was updated from 12.7mm to 12.75, and tolerances on holes in which it sits w [... cut]
closed 2022-02-06 by matthew-reynolds