q-pup/q-pup-mechanical #36: Add Larger Chamfers/Fillets on holes that are on Print bed surfaces

wmmc88, 2022-01-19:
Unless I waste filament and print time on a raft or painstakingly level each time (much pain, very slow), the holes on the bottom can somtimes be a tiiiiinny bit squished. Larger chamfers or fillets would avoid this.

matthew-reynolds, 2022-01-20:
wdym squished? Like undersized?

Which parts have you run into this with so far? Breaking that corner doesn't always sound possible

wmmc88, 2022-01-20:
> wdym squished? Like undersized?
> 
> Which parts have you run into this with so far? Breaking that corner doesn't always sound possible

Cam A hex is the one off the top of my head but I've run into it b4. It's just occasionally the bed might not be perfectly level so the filament on the first layer gets a lil squished cuz it's printing too close